leomcelroy/svg-pcb #9: Trying to reference a pad in snapToPad

kr15h, 2023-01-29:
Hi. Just to understand the code better, I am trying to improve the snapToPad to return a reference, and I have some thoughts.

The way I thought could work would be from withing the `snapToPad` function. 

1. When a pad is hit, one grabs the identifier of the component and tries to find where it is referenced in the codemirror editor.
2. If a variable referencing it is found, construction of the reference to be inserted becomes easy and it can be considered done.
3. If a variable is not found, it is possible to prefix the line `board.add("comp...")` with a variable such as `let comp1 = `. No problem.

So far so clear, but what makes it challenging is that the initial component definitions do not have a name or ID. So I wonder, maybe it makes sense to generate a random one on the fly. 

OK, this is something I realized while writing this and will try, but maybe there is something I am missing.

kr15h, 2023-01-29:
I think I found the names of the footprints, finally, while trying to understand how the astAnalysis work. Got the footprint names there. Let's see if I can figure out how to map them to the PCB footprint instances. Maybe it makes sense to pass the footprint name down the line.

leomcelroy, 2023-01-30:
I've been thinking about this some myself. It seems you're looking in the right places. You can check if the add call is assigned to a variable in astAnalysis and should pass it along as staticInfo.

kr15h, 2023-02-05:
OK, I figured that a good place to play around is when `cursor.name === "VariableDeclaration"` in `astAnalysis`. It took me some time to realize that at this point the editor content is nicely parsed already. My idea at the moment is to get to a point I would be able to pass down the variable name of the object that is returned by `PCB.add`. Then in the snap code, if a variable name for the footprint is found, a reference is inserted, if not, plain coordinates, as it is now.

kr15h, 2023-02-05:
Or maybe it is better as you say, to assign the `add` call and then try to get the variable names from there using [TreeCursor](https://lezer.codemirror.net/docs/ref/#common.TreeCursor) methods.

kr15h, 2023-02-12:
OK, I think I found out a simple way how to do it (d1fe0ff). I started out with hashing the string versions of the PCB.add() calls in astAnalysis, but then realized that it is not going to be easy to create equivalent hashes in pcb.js. There could be a way, but using a simple counter proved to be functional. Unless something dramatic changes in the whole structure, this could remain stable for some time.

closed 2023-03-24 by leomcelroy