leomcelroy/svg-pcb #11: Gerber Export

kr15h, 2023-02-20:
I started looking into the gerber export issue and I see that at this point there is basic functionality that exports 

- Front Copper layer
- Edge Cuts layer
- Drills

It is possible to open them up in KiCad gerber viewer, but if they appear in the way they are expected to appear is another question.

However, it seems that gerber format has many options and that the functionality itself would benefit from having an export options dialog. Maybe we can use this issue to synchronize on the matter. What comes to my mind first is to look at the KiCad gerber export panel and see what could be the bare minimum of options that one could have in order to be able to generate files for, say, three board houses such as 

- JLCPCB
- PCBWay
- AISLER

kr15h, 2023-02-20:
Another thing is that at my lab we use CopperPCB software that supports Gerber X2 format where the drill files are gbr files as well and it would be really nice if it would be able to export those, of course.

leomcelroy, 2023-02-24:
A good objective with this would be to be able to design a Modular Thing in SVG-PCB and order it from a board house without compromises. To do that we would need some other features too. Specifically

- better text on a silkscreen
- solder mask
- complete gerber export

Focusing on the gerber export. Right now it doesn't export all layers of the board design. Something else that is a bit weird is because all the shapes are polylines holes have to be inferred and derived. But that seems to work okay.

kr15h, 2023-03-20:
I started playing around with it and got a modal showing up with options. One set of options is the layer selector and I wonder which structure would be best to use for layers. Is there a built-in set of layers? Are the layers added via code? Does the layer naming convention make sense?

If we look from KiCad perspective, it would make sense to have a few built-in layers at this point:

- F.Cu
- B.Cu
- F.Mask
- B.Mask
- F.Silkscreen (component labels on front side)
- B.Silkscreen (component labels on back side)
- Edge.Cuts (interior, but outline instead of a shape)
- Drills

Should we hard-code these layers in global state? How?

kr15h, 2023-03-20:
I will submit a pull request with initial demo and we can continue the discussion there.

leomcelroy, 2023-03-29:
Quentin and I made a board to test the Gerber export. It's a huge improvement! A couple things came up.

One is we want to verify that the thru-holes are actually plated?

Two is that the back solder mask doesn't get generated.

[image]

Three is that vias also have holes. Which they probably shouldn't. Seen in the stray hole interspersed with the stepsticks below.

[image]

leomcelroy, 2023-03-29:
On closer evaluation the the back solder mask issue may be the JLCPCB gerber viewer.

kr15h, 2023-03-29:
I was using the KiCad Gerber viewer to validate. I also did some testing with JLCPCB, PcbWay and Seeed. They have different viewers and all of them seem buggy in one way or another. But yeah, I am excited too. I came up with some ideas on what classes/datatypes one would need to make the exporting thing easier. Also at this point I feel that basic KiCad export could be something I could do to validate what we talked about earlier.