leomcelroy/svg-pcb #18: Versioning

leomcelroy, 2023-02-24:
Should add versioning metadata to files. When version in file doesn't match the editor version we should alert the user.

At a minimum we should do a tagged release of each version. We may want to host them too.

kr15h, 2023-03-24:
We probably should use semantic versioning https://semver.org/

I would go with a version string in constants.js where there would be 3 constants.

```
const VERSION_MAJOR = 0;
const VERSION_MINOR = 1;
const VERSION_PATCH = 1;
const VERSION_STRING = VERSION_MAJOR.toString() + "." + VERSION_MINOR.toString() + "." + VERSION_PATCH.toString();
```

leomcelroy, 2023-04-12:
version checking added, if version mismatch we should point to a change log or release notes