leomcelroy/svg-pcb #40: Interior cutouts still missing from Gerber

leomcelroy, 2023-04-05:
Despite merging #37 there still seem to be some issues with interior shapes.

Here is a design in SVG-PCB

[image]

And the Gerber in tracespace

[image]

Here is the test file:

```
/* 
@version: v0.0.1

a basic starter design 
*/

/* -- DECLARE_PCB -- */
const board = new PCB();

/* -- DECLARE_COMPONENTS -- */

// constants
const width = 1;
const height = 1;

// board size and shape
let interior = [
  path(
    pt(-0.5, 0.5),
    pt(0.5, 0.5),
    pt(0.5, -0.5),
    pt(-0.5, -0.5),
  ),
  path(
  pt(-0.25, 0.25),
  pt(-0.25, 0),
  pt(-0.1, 0),
  pt(-0.1, 0.25),)
]

board.addShape("interior", interior);

/* -- ADD_COMPONENTS -- */

/* -- RENDER_PCB -- */
const limit0 = pt(-0.55, -0.55);
const limit1 = pt(0.55, 0.55);
const xMin = Math.min(limit0[0], limit1[0]);
const xMax = Math.max(limit0[0], limit1[0]);
const yMin = Math.min(limit0[1], limit1[1]);
const yMax = Math.max(limit0[1], limit1[1]);

renderPCB({
  pcb: board,
  layerColors: {
    "interior": "#002d00ff",
    "B.Cu": "#ff4c007f",
    "F.Cu": "#ff8c00cc",
    "drill": "#ff3300e5",
    //"padLabels": "#ffff99e5",
    "componentLabels": "#00e5e5e5",
  },
  limits: {
    x: [xMin, xMax],
    y: [yMin, yMax]
  },
  mm_per_unit: 25.4
});

```

kr15h, 2023-04-05:
Oh yes I see. Seeing this in KiCad Gerber viewer with this. 
[image]
The solution probably is to double-check if first point is also the last.

kr15h, 2023-04-05:
Now it works in tracespace. There was one more issue with zero values being formatted into void values in Gerber which caused invisible lines in tracespace.
[image]

PR #41 (2023-04-05): Fix gerber close outline
This should close #40. Download Gerber not double-checks if outline is closed.
closed 2023-04-05 by leomcelroy