leomcelroy/svg-pcb #43: Need link to footprint for each component placed

kr15h, 2023-04-07:
There is data lost when components are made out of footprints and some of that data is useful when exporting to KiCad, for example. How about adding a reference to footprint in each board.pcb.components item? One of the things that is getting lost is the pin numbering. Pin labels are fine, but the numbering is more important in KiCad.

leomcelroy, 2023-04-07:
I actually added this yesterday on my dev branch. It's got a few more changes we may not want to merge yet but I can push that one.

kr15h, 2023-04-08:
I figured it out somehow, but the numbering of the pads could be off the way I did it. I can rebase and adjust when you merge that later.

kr15h, 2023-04-08:
I noticed the RefDes, but what would be VERY useful is a reference to the original footprint from each component placed. It would be also perfect if vias would point to an object similar to footprint definition so there would be no need for special case handling. I could try to add this myself unless you have something cooking in your dev branch.

kr15h, 2023-04-08:
> I figured it out somehow, but the numbering of the pads could be off the way I did it. I can rebase and adjust when you merge that later.

And this solution breaks as soon I open up a design with vias.

kr15h, 2023-04-08:
I will try with the new `padShapes`. Maybe it will work.

kr15h, 2023-04-08:
Ah, no, I still would need the pad numbering from the original footprint.

leomcelroy, 2023-04-08:
I'm including this in the component now. It's a bit messy but shouldn't lose anything.

[image]

closed 2023-04-08 by leomcelroy