leomcelroy/svg-pcb #56: KiCad export: handling other layers

ducky64, 2023-05-19:
It seems like currently for KiCad export, only F.Cu and B.Cu layers are exported. But other layers are also important, for example board edge (Edge.Cuts) and silkscreen (F.SilkS or S.Silkscreen depending on convention).

Can these non-copper layers be exported as graphical lines, with the layer name being passed through?

kr15h, 2023-05-22:
Yeah, the `Edge.Cuts` are there, given that there are actual lines in the layer. Just tested with the current live editor and `Edge.Cuts` are there. the `*.Silkscreen` layers show up in KiCad as well, they are just not being populated yet as we have to solve a few things before. Take a look at the screenshot below.

[image]

ducky64, 2023-05-22:
Oh I see, it has to be done with `board.addShape("interior", ...)` and isn't just propagating through lines on Edge.Cuts. Works now!